ros-perception/laser_filters #21: scan_to_cloud_chain produces large-sphere if ~high_fidelity:=false

garaemon, 2015-01-13:
I think this is a bug of laser_geometry...

* high_fidelity:=false
[image]

* high_fidelity:=true
[image]

vrabaud, 2015-01-14:
Actually, it seems that laser_max_range is deprecated in the code but is still used when high_fidelity is false. Do you need that parameter ?

tfoote, 2015-01-14:
This looks like you're spinning a laser relatively quickly. Enabling high_fidelity projection will enable interpolated position of the laser for each point sampled in the scan.  Without that option it transforms all points in a scan at a single point in time. With that flag off, it is much lower computation for the transformation but it trades off the assumption that the motion of the laser is negligable during the duration of the laser scan. 

If you think this is actually a bug. please provide a way to reproduce it such as a relevant bag file and reference material for the enviornment, and the commands to reproduce the display.

garaemon, 2015-01-16:
I uploaded [bag file](www.jsk.t.u-tokyo.ac.jp/~ueda/bags/2015-01-16-23-19-50-scan-to-cloud.bag) and [launch file](https://gist.github.com/garaemon/ecb0929b02e1a6c5ec34).

`/multisense/lidar_scan_cloud_filtered_low_fidelity` and `/multisense/lidar_scan_cloud_filtered_high_fidelity` are the topics where I see this issue.

pijemcolu, 2017-05-17:
I had a very similar issue with a tilting lidar, where I would get a complete sphere at max range of the lidar. I simply fixed it by introducing another range filter dropping points max range -1.