ros-perception/laser_filters #26: laser_filters sources not in distribution.yaml for jade

tkruse, 2015-07-30:
To reproduce:
```
$ rosinstall_generator desktop_full --rosdistro jade --deps --wet-only  --upstream-development
The following repositories with an unknown upstream will be ignored: driver_common, laser_filters, metapackages, rosbag_migration_rule, stage
```

Also see 
https://github.com/ros/filters/issues/11

jonbinney, 2015-07-30:
Ah - looks like I missed something in the release. WIll fix this.

jonbinney, 2015-08-05:
Re-released with source and doc info: https://github.com/ros/rosdistro/pull/9077

Will close this issue after that is merged and the fix is confirmed.

jonbinney, 2015-08-05:
PR was merged; laser_filters now has source repo info in the distribution.yaml. @tkruse I re-ran the command you gave and it now finds laser_filters now. Should be all good.

closed 2015-08-05 by jonbinney