ros-perception/laser_filters #40: pointcloud filters bug find.

asimay, 2016-02-29:
[ERROR] [1456740060.773385720]: Couldn't find filter of type laser_filters/PointCloudFootprintFilter
when use laser_filters/PointCloudFootprintFilter filter.
find the reason: scan_to_cloud_filter_chain.cpp listen the Pointcloud2 message, but point_cloud_footprint_filter.h send & plugin register with Pointcloud message. message is not coincide with each, compile is no error, but run error.

jonbinney, 2016-02-29:
Sounds like `laser_filters/PointCloudFootprintFilter` was never updated to use the PointCloud2 message type. Shouldn't be too hard to fix; any chance you could submit another PR for this? :-)

jonbinney, 2016-02-29:
Actually it looks like that filter is deprecated. Can you use "footprint filter" instead? 

https://github.com/ros-perception/laser_filters/blob/indigo-devel/include/laser_filters/footprint_filter.h

It operates directly on laser scans.

asimay, 2016-03-01:
yes. footprint filter maybe able to work. because it listens the laserscan massage, not Pointcloud massage, i just want to test point cloud filter. so maybe this project still need more development about newer's filters code.