ros-perception/laser_filters #54: laser_filters/LaserArrayFilter Bug

spuetz, 2017-07-19:
Defining only the param range_filter_chain causes an error: "process has died!" Right now you need both: range_filter_chain, intensity_filter_chain.

Using the MedianFilterFloat and not MultiChannelMedianFilterFloat is also not possible. Couldn't find filter of type filters/MedianFilterFloat. MultiChannelMedianFilterFloat is working.

jonbinney, 2017-07-19:
What version of ros are you on? What are you launching (text of config files would be useful). Are you using this package from source or from debs?

spuetz, 2017-07-19:
We use kinetic, with the Debian package. For example, the following config (without `intensity_filter_chain`) does not work:

```
scan_filter_chain:
- type: laser_filters/LaserArrayFilter
  name: laser_median_filter
  params: 
    range_filter_chain:
      - name: median_5
        type: filters/MultiChannelMedianFilterFloat 
        params:
          number_of_observations: 5
```

jonbinney, 2017-07-19:
I haven't been able to reproduce your problem yet. Does it crash immediately, or only when you receive laser scans? Any chance you can provide a gdb backtrace? Are there any errors other than "process has died!"? Can you copy/paste the entire terminal output?