ros-perception/laser_filters #64: Release into Melodic

clalancette, 2018-04-10:
It looks like all of the dependencies for this package are available in Melodic, so it would be great to get this released.  Thanks in advance.

jonbinney, 2018-04-11:
@clalancette I'll try and get to this tonight.

jonbinney, 2018-04-11:
Argh, didn't finish. Will try again tomorrow. I want to build this first, but need to setup docker since I don't have an 18.04 machine handy.

clalancette, 2018-04-11:
Great, thanks for taking a look.

Not sure if it helps at all, but I've been pretty happy using LXC to do my development on bionic: https://help.ubuntu.com/lts/serverguide/lxc.html .  You'd basically do:

```
sudo apt-get install lxc
sudo lxc-create -t download -n u1 -- --dist ubuntu --release bionic --arch amd64
```

And then you can get into the container with:

```
sudo lxc-attach -n u1
```

jonbinney, 2018-04-12:
Got my container up and running, but didn't have time to test. Went ahead and released anyway; maybe we get lucky and the build works? If not I'll investigate further.

jonbinney, 2018-04-12:
It built! Hooray!

closed 2018-04-12 by jonbinney