ros-perception/laser_filters #78: Inverse Box Filter or ROI filter

Jailander, 2019-06-20:
Hi,

I was wondering why there isn't any **inverse** Box Filter?
I mean a filter that filters out all measurements outside of the box or **region of interest**, it seems that it could be done just as a copy of the Box filter plugin but changing this [line](https://github.com/ros-perception/laser_filters/blob/indigo-devel/src/box_filter.cpp#L177)
to: `if(!inBox(point)){`.
 
Is there any way of already doing this that I am missing, or is my proposed solution a contribution worth making?

jonbinney, 2019-06-20:
@Jailander i think it would be even easier to just add this feature to the existing box filter by adding an "invert_selection" parameter to the filter which negates that check. And if that param defaults to false, then adding it won't break any existing use cases. If you're interested, go ahead and make a PR.

Jailander, 2019-06-21:
Oh that is a good idea :+1: , I'll make the PR when I test it

Jailander, 2019-06-25:
Hi I made the PR as promised it works like a charm for me :+1: thank you for your suggestion

jonbinney, 2019-06-25:
Thanks @Jailander ! I'll take a look!

Jailander, 2019-07-02:
I saw the PR has been merged, thank you will close this issue :)

closed 2019-07-02 by Jailander