ros-perception/laser_filters #81: boxFilter invert param not usable in melodic

athish-t, 2019-09-02:
I am using ROS Melodic and I'm trying to use the `invert` param in box_filters which was added in #79 
I upgraded the `laser_filters` package by `sudo apt-get update && sudo apt-get install ros-melodic-laser-filters`, even tried uninstalling and installing again.
But I'm still not able to use the parameter. Perhaps the debian distribution has not been updated to the latest version yet?

jonbinney, 2019-09-03:
Yeah, there hasn't been a release since that PR. I'll try to get one done in the next week.

athish-t, 2019-11-06:
> Yeah, there hasn't been a release since that PR. I'll try to get one done in the next week.

Hi, @jonbinney ..any updates on this?

jonbinney, 2019-11-07:
@athish-t thanks for the reminder! I just did the release: https://github.com/ros/rosdistro/pull/22909

It will take a few days to make it into the "testing" apt repository, and then will get into the "public" apt repo after the next sync (tends to happen every few weeks)

closed 2019-11-07 by jonbinney