ros-perception/laser_filters #86: Box filter not filtering noise

gstavrinos, 2019-11-21:
I am using the box filter with a Hokuyo  UST-10LX (10m maximum measuring distance). Sometimes the sensor returns values with a much larger value than the maximum one (>65m). Of course, this is just noise and should be disregarded. The problem is, that **I need** to use the range filter complementary to the box filter in order to have those "garbage" values disappear, because box filter alone, won't do that.

It seems to be a bug, but skimming through the box filter code, did not bring up anything out of the ordinary. Maybe I am missing something obvious. Can anyone enlighten me, please?

gstavrinos, 2019-11-21:
In order to make my problem clearer, here is a quick example.

This is part of the ranges vector with both the box filter and ranges filter:

```
[...] nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, inf, 1.062999963760376, 
1.0520000457763672, 1.0410000085830688, 1.0549999475479126, 1.0470000505447388, 
1.0440000295639038, 1.0399999618530273, 1.0399999618530273, 1.0440000295639038, 
1.0440000295639038, 1.0360000133514404, 1.0360000133514404, 1.034000039100647, 
1.0290000438690186, 1.0299999713897705, 1.031000018119812, 1.0360000133514404, 
1.0449999570846558, 1.0509999990463257, 1.0520000457763672, 1.0509999990463257, 
1.0520000457763672, 1.059000015258789, 1.0670000314712524, 1.0679999589920044, 
1.059000015258789, 1.0729999542236328, inf, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, 
nan, nan, nan, nan, inf, 1.4559999704360962, 1.4479999542236328, 1.4459999799728394, 
1.4609999656677246, 1.4539999961853027, 1.4479999542236328, 1.4459999799728394, 
1.4479999542236328, 1.4479999542236328, 1.4470000267028809, 1.444000005722046, 
1.437999963760376, 1.437999963760376, 1.4390000104904175, 1.440999984741211, 
1.4429999589920044, 1.440000057220459, 1.4570000171661377, 1.444000005722046, 
1.4539999961853027, inf, [...]
```

And this is the same part without the range filter:

```
[...] nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, 65.53299713134766, 1.062999963760376, 
1.0520000457763672, 1.0410000085830688, 1.0549999475479126, 1.0470000505447388, 
1.0440000295639038, 1.0399999618530273, 1.0399999618530273, 1.0440000295639038, 
1.0440000295639038, 1.0360000133514404, 1.0360000133514404, 1.034000039100647, 
1.0290000438690186, 1.0299999713897705, 1.031000018119812, 1.0360000133514404, 
1.0449999570846558, 1.0509999990463257, 1.0520000457763672, 1.0509999990463257, 
1.0520000457763672, 1.059000015258789, 1.0670000314712524, 1.0679999589920044, 
1.059000015258789, 1.0729999542236328, 65.53299713134766, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, 
nan, nan, nan, nan, 65.53299713134766, 1.4559999704360962, 1.4479999542236328, 1.4459999799728394, 
1.4609999656677246, 1.4539999961853027, 1.4479999542236328, 1.4459999799728394, 
1.4479999542236328, 1.4479999542236328, 1.4470000267028809, 1.444000005722046, 
1.437999963760376, 1.437999963760376, 1.4390000104904175, 1.440999984741211, 
1.4429999589920044, 1.440000057220459, 1.4570000171661377, 1.444000005722046, 
1.4539999961853027, 65.53299713134766, [...]
```

Notice that in the case of the box filter without the range filter, the inf values are the noisy 65.53299713134766 value. I would expect the box filter to replace these values with nan.

nlimpert, 2020-04-07:
I had a very similar problem with SICK lasers showing sensor noise with points very similar to yours.
In my situation the noisy beams were at least within the sensor's maximum range but they caused quite some trouble during navigation (phantom obstacles).
So I tried to configure the `ScanShadowFilter` to my needs but didn't really succeed to get a robust and reliable setup working. Then I ended up implementing my own laser filter which was also recently merged in https://github.com/ros-perception/laser_filters/pull/93 and the new repo's release might also be merged at https://github.com/ros/rosdistro/pull/24413.

Maybe that helps you?

gstavrinos, 2020-04-10:
Hey @nlimpert , thanks for the heads up! As I said in my original post, I hacked my way through the problem by using the range filter in addition to the box filter. I will definitely keep in mind your new filter though. Thanks again!