ros-perception/laser_filters #87: LaserScanRangeFilter not working.

kjitender469, 2019-11-29:
My launch file is:

<launch>
  <node pkg="laser_filters" type="scan_to_scan_filter_chain"
      name="laser_filter">
  
    <rosparam command="load" file="$(find rplidar_filter)/rplidar_range_filter_config.yaml" />
    <remap from="scan_filtered" to="range_scan" />
    
  </node>
</launch>

My rplidar_range_filter_config.yaml is : 

scan_filter_chain:
- name: range_filter
  type: laser_filters/LaserScanRangeFilter
  params:
    use_message_range_limits: false   
    lower_threshold: 0.5              
    upper_threshold: 1.0              
    lower_replacement_value: -.inf    
    upper_replacement_value: .inf

JohnTGZ, 2022-01-29:
Hi there, could you highlight the problems you are facing and provide some data?

fantalukas, 2022-10-21:
I was facing the same issue (LaserScanRangeFilter was not working). But I managed to fix this issue (problem was in the naming of the node). I will describe my scenario and a solution here. Maybe it could help you as well.

**Expected behavior:**
Ranges of scan data are filtered as requested in YAML file.

**Current behavior:**
Ranges of scan data are not filtered.

**Scenario:**
I used example case of range filter (current commit: https://github.com/ros-perception/laser_filters/commit/cf27b3654f5e998d21aa6aeb72e37c8663ce422b launch file: https://github.com/ros-perception/laser_filters/blob/ros2/examples/range_filter_example.launch.py,  yaml file: https://github.com/ros-perception/laser_filters/blob/ros2/examples/range_filter_example.yaml) and slightly modify that example.

More specifically, I slightly modified only launch file range_filter_example.launch.py as follows (the only change is adding name of node compare to example):

[image]

**Comparing my and example versions:**
MY modified version of example **is NOT working**. More specifically, _update_ and _configure_ methods are not called in _LaserScanRangeFilter_ class (tested by logging). So the ranges of scan data are not filtered.
EXAMPLE version **is working**. Ranges of scan data are filtered as requested in yaml file.

**Solution:**
The only difference is that I added the name of the node to the launch file (_name_ parameter).
After removing the name of node. It is working.

So the problem is in naming of the node.