ros-perception/laser_filters #103: ROS 2 pcl_ros dependencies does not exist

Patrick-Lascombe, 2021-01-29:
Hello everyone,

In the ROS2 branch, the point_cloud_footprint_filter include "pcl_ros/transforms.hpp". But pcl_ros does not exist anymore in ros2 foxy. So the package does not pass the compilation.

Is anyone having the same problem as me ? Or do I simply comment the include and plugin lib export in pointcloud_filters to pass the compilation ?

Thanks for your help

jonbinney, 2021-01-29:
Good question. I haven't actually used this package in foxy. @hangst did you get things working on foxy?

hangst, 2021-01-30:
The `pcl_ros` package does has a `foxy-devel` branch. I believe it is not released as binary, so you will have to compile from [sources](https://github.com/ros-perception/perception_pcl/tree/foxy-devel/pcl_ros) manually. 

Alternatively, as you mentioned, if you are not planning to use `scan_to_cloud_filter_chain`, you may modify `CMakelists.txt` and `package.xml` to compile without `pcl_ros`. 
```cmake
ament_auto_add_library(laser_scan_filters SHARED src/laser_scan_filters.cpp)
# ament_auto_add_library(pointcloud_filters SHARED src/pointcloud_filters.cpp)

set(FILTER_CHAINS
    # scan_to_cloud_filter_chain
    scan_to_scan_filter_chain
    # generic_laser_filter_node
)
```
```xml
<depend>rostime</depend>
<!--<depend>pcl_ros</depend>-->
<depend>rclcpp_lifecycle</depend>
```

Patrick-Lascombe, 2021-02-01:
Alright I'll keep it commented for now 
Thanks for your help

closed 2021-02-01 by Patrick-Lascombe