ros-perception/laser_filters #108: Bug: package build fails, hpp files not found

AliaChe, 2021-03-09:
Hello, 

I'm trying to test this package using ros foxy. I have clone this repos and the filters repos [https://github.com/ros/filters/tree/ros2](url).
I verified that all the dependencies are installed in my pc, except for pcl_ros and rostime. I commented the dependencies to these two packages as recommended in #103.

However, I still can't build the package successfully, the error is :
` colcon build --packages-select laser_filters                                                                                                                                                      130 ↵
Starting >>> laser_filters
--- stderr: laser_filters                             
/home/blackdrones/blackdrones_ws/src/laser_filters/test/test_scan_filter_chain.cpp:31:10: fatal error: filters/filter_chain.hpp: No such file or directory
   31 | #include <filters/filter_chain.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/test_scan_filter_chain.dir/build.make:63: CMakeFiles/test_scan_filter_chain.dir/test/test_scan_filter_chain.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:104: CMakeFiles/test_scan_filter_chain.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/blackdrones/blackdrones_ws/src/laser_filters/src/laser_scan_filters.cpp:30:
/home/blackdrones/blackdrones_ws/src/laser_filters/include/laser_filters/median_filter.h:42:10: fatal error: filters/median.hpp: No such file or directory
   42 | #include "filters/median.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/laser_scan_filters.dir/build.make:63: CMakeFiles/laser_scan_filters.dir/src/laser_scan_filters.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:158: CMakeFiles/laser_scan_filters.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< laser_filters [0.67s, exited with code 2]

Summary: 0 packages finished [0.87s]
  1 package failed: laser_filters
  1 package had stderr output: laser_filters
`
Thanks for your help

tuxnese, 2021-03-09:
I have been able to work around it by adding set(BUILD_TESTING false) at the beginning of the test section of the CMakeLists.txt file.

AliaChe, 2021-03-10:
This doesn't work for me, also if the test failed doesn't it mean that the execution will fail?

tuxnese, 2021-03-11:
The filters work fine in my case. I think that the test files have includes that are not needed by the rest of the package.

AliaChe, 2021-03-11:
Thank you for your reply @tuxnese . I fixed this error by adding 
 " ament_export_include_directories(include) "  in the CMakeList.txt of the package filters, then by desactivating the test as you suggested. Now I'm running into another error when trying to use the filters
```
[scan_to_scan_filter_chain-13] [INFO] [1615475503.537921136] [scan_to_scan_filter_chain]: LaserArrayFilter cleaning and reallocating due to larger scan size
[scan_to_scan_filter_chain-13] [WARN] [1615475503.541828472] [pluginlib.ClassLoader]: given plugin name 'lib/libmedian' should be '/libmedian' for better portability
[scan_to_scan_filter_chain-13] [FATAL] [1615475503.542328199] [scan_to_scan_filter_chain]: Could not load library for filters/MultiChannelMedianFilterFloat: Could not find library corresponding to plugin filters/MultiChannelMedianFilterFloat. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [scan_to_scan_filter_chain-13]: process has died [pid 25284, exit code -11, cmd '/home/blackdrones/blackdrones_ws/install/laser_filters/lib/laser_filters/scan_to_scan_filter_chain --ros-args --params-file /home/blackdrones/blackdrones_ws/install/blackdrones_control/share/blackdrones_control/config/params.yaml -r scan:=/laserscan/out'].

```

Any ideas?

jonbinney, 2021-03-12:
Looks like there is a PR on the filters repo to add the ament_export_include_directories: https://github.com/ros/filters/pull/46 . I commented on that issue to see if we can get it merged.

@AliaChe could you open a separate issue for the new bug you're seeing?

AliaChe, 2021-03-12:
Great news for the PR, I was going to open one. 

okay, I'll open a separate issue for the new bug. Thx!

AliaChe, 2021-03-16:
Issue solved

jonbinney, 2021-03-19:
I'm going to leave this issue open until the filters PR gets merged, just so we have a place to track the problem here.

jonbinney, 2021-06-20:
filters PR hasn't been merged yet; I've bumped the maintainer.

jonbinney, 2021-08-04:
Fix was merged upstream and release.

closed 2021-08-04 by jonbinney