ros-perception/laser_filters #109: LaserArrayFilter bug

AliaChe, 2021-03-12:
Hello,

I'm trying to test this package using ros foxy.  I've got an error when using the LaserArrayFilter, which claims to not find the necessary librairies from the "filters" package. 

Here is the error:

```
[scan_to_scan_filter_chain-13] [INFO] [1615475503.537921136] [scan_to_scan_filter_chain]: LaserArrayFilter cleaning and reallocating due to larger scan size
[scan_to_scan_filter_chain-13] [WARN] [1615475503.541828472] [pluginlib.ClassLoader]: given plugin name 'lib/libmedian' should be '/libmedian' for better portability
[scan_to_scan_filter_chain-13] [FATAL] [1615475503.542328199] [scan_to_scan_filter_chain]: Could not load library for filters/MultiChannelMedianFilterFloat: Could not find library corresponding to plugin filters/MultiChannelMedianFilterFloat. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [scan_to_scan_filter_chain-13]: process has died [pid 25284, exit code -11, cmd '/home/blackdrones/blackdrones_ws/install/laser_filters/lib/laser_filters/scan_to_scan_filter_chain --ros-args --params-file /home/blackdrones/blackdrones_ws/install/blackdrones_control/share/blackdrones_control/config/params.yaml -r scan:=/laserscan/out'].
```

jonbinney, 2021-03-12:
@hangst did you have this problem when you used laser filters on foxy?

jonbinney, 2021-03-13:
Renamed for issue to be more specific. I suspect this is the solution: https://github.com/ros/filters/pull/44

@AliaChe could you cherry pick the commit from that pull request and see if it fixes the problem?

hangst, 2021-03-13:
> @hangst did you have this problem when you used laser filters on foxy?

Yes, I faced this issue with released version of `ros/filters` package. 

Following the warning messages from `pluginlib.ClassLoader`, I managed to fix the issue by modifying `default_plugins.xml`, and therefore proposed PR at ros/filters#44.

AliaChe, 2021-03-15:
This solved the issue. Thanks!

AliaChe, 2021-03-16:
Issue solved using ros/filters #44

AliaChe, 2021-03-16:
Issue solved

closed 2021-03-16 by AliaChe