ros-perception/laser_filters #112: How use LaserScanAngularBoundsFilter to select the short section

Teo-Cardoso, 2021-04-09:
I have a Lidar that its front it is the angle_min and angle_max, and I would like to select just the section in yellow ( image ). However, the parameters lower_angle and upper_angle can't select this ( or I don't know how to do it ). I just can select between -pi -> 0 -> pi.

[image]

How can I configure this to make what I want? 

I don't know if I make myself clear, but any questions about my question just ask.

closed 2021-04-22 by Teo-Cardoso

rodrigogic, 2021-07-20:
Hi @Teo-Cardoso, I'm facing the same issue. Did you find any solutions for this?

Teo-Cardoso, 2021-07-20:
> Hi @Teo-Cardoso, I'm facing the same issue. Did you find any solutions for this?
@rodrigogic 

Yes, now I'm using the `LaserScanAngularBoundsFilterInPlace` instead. 
This one removes the points inside of the interval instead of outside ( as `LaserScanAngularBoundsFilter` does )
I select from -1.5707 to 1.5707 and these points are removed, remaining only the -1.507<-->-3.14/3.14<-->1.5707

rodrigogic, 2021-07-23:
@Teo-Cardoso thanks a lot! I'll try the same here.