ros-perception/laser_filters #143: How to compile with Eloquent (Ubuntu18.04)?

hiroyuki-s1, 2021-11-15:
How to compile with Eloquent (Ubuntu18.04)?
I have a problem with Boost references.
I modified the CMakeLists.txt file as follows, but it does not compile.

CMakeLists.txt

```
cmake_minimum_required(VERSION 3.5)
project(laser_filters)

##############################################################################
# Find dependencies
##############################################################################

find_package(Boost REQUIRED)
find_package(ament_cmake_auto REQUIRED)
ament_auto_find_build_dependencies()

include_directories(laser_scan_filters ${Boost_INCLUDE_DIRS})

##############################################################################
# Build
##############################################################################

ament_auto_add_library(laser_scan_filters SHARED src/laser_scan_filters.cpp)
link_directories(laser_scan_filters ${Boost_LIBRARY_DIRS})
target_link_libraries(laser_scan_filters ${Boost_LIBRARIES})

set(FILTER_CHAINS
    scan_to_cloud_filter_chain
    scan_to_scan_filter_chain
    generic_laser_filter_node
)

set(LIBS
-lboost_system
)

foreach(FILTER_CHAIN ${FILTER_CHAINS})
    ament_auto_add_executable(${FILTER_CHAIN} src/${FILTER_CHAIN}.cpp)
endforeach()
##############################################################################
# Install
##############################################################################

pluginlib_export_plugin_description_file(filters laser_filters_plugins.xml)
ament_auto_package(INSTALL_TO_SHARE examples)

##############################################################################
# Test
##############################################################################

if(BUILD_TESTING)
    find_package(ament_cmake_gtest REQUIRED)
    set(TEST_NAME test_scan_filter_chain)
    set(RESULT_FILENAME ${AMENT_TEST_RESULTS_DIR}/${PROJECT_NAME}/${TEST_NAME}.gtest.xml)
    ament_add_gtest_executable(${TEST_NAME} test/${TEST_NAME}.cpp)
    ament_target_dependencies(${TEST_NAME} filters pluginlib rclcpp)
    ament_add_test(
        ${TEST_NAME}
        COMMAND 
            $<TARGET_FILE:${TEST_NAME}>
            --ros-args --params-file ${PROJECT_SOURCE_DIR}/test/${TEST_NAME}.yaml
            --gtest_output=xml:${RESULT_FILENAME}
        RESULT_FILE ${RESULT_FILENAME}
    )
endif()

```

result
```
Starting >>> laser_filters
--- stderr: laser_filters                             
liblaser_scan_filters.so: undefined reference to `boost::system::system_category()
liblaser_scan_filters.so: undefined reference to `boost::system::generic_category()'.
collect2: error: ld returned 1 exit status
make[2]: *** [scan_to_scan_filter_chain] Error 1
make[1]: *** [CMakeFiles/scan_to_scan_filter_chain.dir/all] Error 2
make[1]: *** Waiting for an incomplete job ....
liblaser_scan_filters.so: undefined reference to `boost::system::system_category()'.
liblaser_scan_filters.so: undefined reference to `boost::system::generic_category()'.
collect2: error: ld returned 1 exit status
make[2]: *** [generic_laser_filter_node] Error 1
make[1]: *** [CMakeFiles/generic_laser_filter_node.dir/all] Error 2
liblaser_scan_filters.so: undefined reference to `boost::system::system_category()'.
liblaser_scan_filters.so: undefined reference to `boost::system::generic_category()'.
collect2: error: ld returned 1 exit status
make[2]: *** [scan_to_cloud_filter_chain] Error 1
make[1]: *** [CMakeFiles/scan_to_cloud_filter_chain.dir/all] Error 2
make: *** [all] Error 2
```

jonbinney, 2021-11-15:
Which branch of this repo are you using? What ros distro are you on?

hiroyuki-s1, 2021-11-16:
I'm trying to use the ros2 branch with Eloquent ( Ubuntu18.04 ).

jonbinney, 2021-11-17:
I'm not sure what's going on there.... I'm on foxy/galactic/rolling on 20.04, and I'm not seeing this. If I have time I'll try and look a bit deeper, but it's been a while since eloquent went end-of-life.