ros-perception/laser_filters #147: Is there an inverted Box-Filter?

NikolasE, 2022-01-04:
I would like to use the box-filter to remove everything but(!) the points in front of the robot, but in the current implementation, the points in the box are deleted:

```
      if (inBox(point))
        {
          output_scan.ranges[*iter_i] = std::numeric_limits<float>::quiet_NaN();
        } 
```

Is there already in inverted version? (otherwise I'll create a small PR with an additional "keep_box_points"-parameter that defaults to false)

JohnTGZ, 2022-01-29:
May I know which branch are you currently on? I think on the "noetic-devel" branch, there should already be a parameter for box_filter to invert "removing" the points. 
You can also find the source code that corresponds to this implementation [here](https://github.com/ros-perception/laser_filters/blob/c25f2f738b33f322ffb3bb52f2dc01d74688b233/src/box_filter.cpp#L185):
```
    if(!invert_filter){
      if(inBox(point)){
        output_scan.ranges[index] = std::numeric_limits<float>::quiet_NaN();
      }
    }
    else{
      if(!inBox(point)){
        output_scan.ranges[index] = std::numeric_limits<float>::quiet_NaN();
      }
    }
```

jonbinney, 2022-01-30:
I think that @NikolasE is on the ros2 branch; we've had some discussion about porting the invert param to that branch here: https://github.com/ros-perception/laser_filters/pull/148

closed 2022-11-03 by NikolasE