ros-perception/laser_filters #152: Laser_filter launch file does not do anything when launched, base_scan might be issue?

DistroIII, 2022-01-27:
I'm using a turtlebot3 which publishes all its LiDAR sensor data to the ```/scan``` topic. I need to use the laser_filters package to clean up the data. I installed the package and this is the launch file I made, according to the example in the wiki to try to use this package.

```
<launch>
  <node pkg="laser_filters" type="scan_to_scan_filter_chain"
      name="laser_filter">
    <rosparam command="load" file="$(find test)/my_laser_config.yaml"/>
    <remap from="scan" to="base_scan"/>
  </node>
</launch>
```
I roslaunch the file above and it literally does nothing, I ```rostopic echo /base_scan``` and no information is being sent to that topic. What do I do? I'm at my wits end. when I delete this line ```   <remap from="scan" to="base_scan"/> ```, I get an error.
when I change "base_scan" to "$(arg scan_topic)", I get an error. Please I need help. What am I doing wrong?

jonbinney, 2022-01-30:
This kind of high level debugging question would be better for ROS answers. Could you ask it there? https://answers.ros.org/questions/

closed 2022-01-30 by jonbinney

DistroIII, 2022-01-30:
@jonbinney I did so [here](https://answers.ros.org/question/395232/turtlebot3-lidar-laser_filter-package-error-reason-new-node-registered-with-same-name/?answer=395343#post-id-395343).