ros-perception/laser_filters #176: Release a foxy build

neochaos42, 2023-03-18:


jonbinney, 2023-03-18:
Will do!

jonbinney, 2023-03-18:
Released into foxy; rosdistro PR is here: https://github.com/ros/rosdistro/pull/36518

Looks like foxy hits end-of-life in may; you might want to update to humble. I went ahead and also made new releases into humble and rolling:
Rolling: https://github.com/ros/rosdistro/pull/36516
Humble: https://github.com/ros/rosdistro/pull/36517

neochaos42, 2023-03-27:
https://index.ros.org/p/laser_filters/github-ros-perception-laser_filters/#foxy still is empty

jonbinney, 2023-03-28:
Not sure how index.ros.org works, but that page says it was last updated in december, so I doubt it would have it?

It looks like laser filters built for foxy: https://build.ros2.org/view/Fbin_uF64/job/Fbin_uF64__laser_filters__ubuntu_focal_amd64__binary/lastBuild/console

I'm guessing it didn't quite make it in time for the last foxy sync, which was a week ago: https://discourse.ros.org/t/new-packages-and-patch-release-for-foxy-fitzroy-2023-03-22/30489

In which case i think we have to wait for the next sync?