ros-perception/laser_filters #182: Can't run laserfilters on two different scan topics.

JosefGst, 2023-07-14:
I would like to run two instances of laser_filters, one on '/front/scan' and the second on '/back/scan'. But when I define namespace like: 
```
Node(
  package="laser_filters",
  executable="scan_to_scan_filter_chain",
  namespace="front",
  parameters=[laser_filter, {'use_sim_time': LaunchConfiguration('use_sim_time')}],
),
```
the filter will do nothing but just relay an unfiltered result. Everything works great when not using namespace. Is there a method so it can filter on two topics?

jonbinney, 2023-07-14:
So you're launching two separate nodes, one to handle filtering the front scans, and one to filter the back scans, right? Could you post the output of `ros2 node info <node_name>` for each of those two nodes while they are running?

JosefGst, 2023-07-18:
Thank you for your fast response! I launch the front laser filter with namespace and it breaks the filter.
```
# LASER FILTER FRONT
Node(
    package="laser_filters",
    executable="scan_to_scan_filter_chain",
    namespace="front",
    parameters=[laser_filter, {'use_sim_time': LaunchConfiguration('use_sim_time')}],
),
# LASER FILTER BACK
Node(
    package="laser_filters",
    executable="scan_to_scan_filter_chain",
    # namespace="back",
    parameters=[laser_filter, {'use_sim_time': LaunchConfiguration('use_sim_time')}],
    remappings=[
        ('/scan', '/back/scan'),
        ('/scan_filtered', '/back/scan_filtered'),
    ]
),
```
But laser back can work normally because "namespace" is commented out. 
Following I got form the node info command:
```
ros2 node info /front/scan_to_scan_filter_chain 
/front/scan_to_scan_filter_chain
  Subscribers:
    /clock: rosgraph_msgs/msg/Clock
    /front/scan: sensor_msgs/msg/LaserScan
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /front/scan_filtered: sensor_msgs/msg/LaserScan
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Service Servers:
    /front/scan_to_scan_filter_chain/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /front/scan_to_scan_filter_chain/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /front/scan_to_scan_filter_chain/get_parameters: rcl_interfaces/srv/GetParameters
    /front/scan_to_scan_filter_chain/list_parameters: rcl_interfaces/srv/ListParameters
    /front/scan_to_scan_filter_chain/set_parameters: rcl_interfaces/srv/SetParameters
    /front/scan_to_scan_filter_chain/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
  Service Clients:

  Action Servers:

  Action Clients:

```
```
ros2 node info /scan_to_scan_filter_chain 
/scan_to_scan_filter_chain
  Subscribers:
    /back/scan: sensor_msgs/msg/LaserScan
    /clock: rosgraph_msgs/msg/Clock
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /back/scan_filtered: sensor_msgs/msg/LaserScan
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Service Servers:
    /scan_to_scan_filter_chain/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /scan_to_scan_filter_chain/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /scan_to_scan_filter_chain/get_parameters: rcl_interfaces/srv/GetParameters
    /scan_to_scan_filter_chain/list_parameters: rcl_interfaces/srv/ListParameters
    /scan_to_scan_filter_chain/set_parameters: rcl_interfaces/srv/SetParameters
    /scan_to_scan_filter_chain/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
  Service Clients:

  Action Servers:

  Action Clients:
```
The "front/scan_filtered" topic is still published but it is unfilterd, just an exact copy of the input "/front/scan". If you need any further information please let me know.

jonbinney, 2023-07-18:
I'd guess that the parameters aren't getting loaded properly into the namespace for the front node. What is the value of the variable `laser_filter` in the parameter list? You may need to use the `/**` syntax so that ros2 launch will put the parameters in the correct namespace for each node.

JosefGst, 2023-07-19:
Awesome! Thank you for your help! It was indeed about the parameter namespace. I changed the yaml file into:
```
namespace1:
  scan_to_scan_filter_chain:
    ros__parameters:
      filter1:
       ...
namespace2:
  scan_to_scan_filter_chain:
    ros__parameters:
      filter1:
        ...
```
and the namespacing through the launch file is working agian :+1:

closed 2023-07-19 by JosefGst