ros-perception/laser_filters #193: Extend footprint_filter for non circular robots

jcarlosgm30, 2024-05-22:
Hi!

The current implementation of footprint_filter is only for circular robots. It would be very useful if we extend the filter to be used by non-circular robots including a footprint in a similar format as NAV2.

what do you think about this? @jonbinney

jonbinney, 2024-05-23:
Yes, definitely useful! I think the [polygon filter](https://github.com/ros-perception/laser_filters/blob/noetic-devel/include/laser_filters/polygon_filter.h) does that in ROS1, and there is actually a PR by @berend-kupers to port it (and other filters) to ROS2: https://github.com/ros-perception/laser_filters/pull/189/files . Its a big PR and I haven't had time to review it - if you have some time and can do a review that would help move it along!

AntoBrandi, 2025-01-08:
I see that https://github.com/ros-perception/laser_filters/pull/189 has been merged 🎉
Will it be released to humble?