ros-perception/laser_filters #214: Creating multiple nodes

rafal-gorecki, 2025-03-28:
When I set name in launch file, I get the following error

```bash
[scan_to_scan_filter_chain-5] [WARN] [1743186960.346687424] [rcl.logging_rosout]: Publisher already registered for node name: 'laser_filter'. If this is due to multiple nodes with the same name then all logs for the logger named 'laser_filter' will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
```

```python
    laser_filter_node = Node(
        package="laser_filters",
        executable="scan_to_scan_filter_chain",
        name="laser_filter", # Removing this fix the problem
        parameters=[laser_filter_config],
    )
```

jonbinney, 2025-03-29:
Sounds like you're launching another node with the same node somewhere .....

rafal-gorecki, 2025-03-31:
I understand that it may sound like that, but it looks more like an error related to the use of parameters (the name has nothing to do with it, and the error for another name did not appear because I did not change the node name in the file with parameters)

Try to reproduce this problem on ROS jazzy with the file `~/params.yaml`

```yaml
---
/**/laser_filter:
  ros__parameters:
    filter1:
      name: box_filter
      type: laser_filters/LaserScanBoxFilter
      params:
        box_frame: base_link
        max_x: 0.1
        min_x: -0.17
        max_y: 0.12
        min_y: -0.12
        max_z: 0.2
        min_z: 0.0
        invert: false
```

and the command: 

```bash
ros2 run laser_filters scan_to_scan_filter_chain --ros-args --remap __node:=laser_filter --params-file ~/laser_filter.yaml
```

jonbinney, 2025-04-02:
I was able to reproduce this on jazzy/noble, so I retract my sarcastic comment about you running multiple nodes with the same name....

I strongly suspect that this is the result of some filters (including box_filter) starting their own nodes. If i remember correctly, this was necessary when porting those filters from ROS1 because the FilterBase class didn't provide any way to setup parameter callbacks for writeable parameters. The parameter handling with the filters package has been improved since then, so we may be able to remove the LifeCycleNode from the various laser filters. If i remove the `-remap __node:=laser_filter` the error goes away, so I suspect that the remapping is changing the name of both the top level scan_to_scan_filter_chain node and also the LifeCycleNode within the box filter, causing the collision.

I don't have time to do further debugging on this until next week - if you've got time a PR fixing this would speed things up :-)

jonbinney, 2025-04-09:
After https://github.com/ros/filters/pull/83 gets merged into filters, it should be possible to remove the extra nodes from the laser filters that need dynamically updated parameters, which I think will solve this issue.

rafal-gorecki, 2025-04-09:
Thank you for the information and taking care of the matter!